bagges/deye-esp32-bridge #3: Support for Pace BMS

engel75, 2023-07-24:
Hi,

I really like to test the esp32-bridge with my Deye inverter and my battery. My battery uses a Pace BMS  (P16s200A) and syssi just pushed his esp32 pace BMS config:
https://github.com/syssi/esphome-pace-bms

I never merged different modbus contollers in one configuration. Is it possible to add Pace BMS support to the bridge?

bagges, 2023-07-24:
Hi, I just checked syssi's repo. Looks like Pace has the same pinout as Seplos so it should work. I cannot test it as I do not have a Pace BMS. You just need to adapt the esphome config.

bagges, 2023-07-27:
I added an example for PACE BMS: https://github.com/bagges/deye-esp32-bridge/blob/main/deye-esp32-bridge-pace.yaml

You need to enable modbus first on Pace BMS. refer to: https://github.com/syssi/esphome-pace-bms#esphome-pace-bms

engel75, 2023-08-03:
Thank you for adding this example!

I am almost finished with the PCB board, but I would like to clarify some questions I have regarding your wiring description:

"Connect any 7-24V DC source to the screw headers (do not use CN2 PINS 7+8 from deye inverter, they only work for about 100mA). Make sure that JP2 is closed."

screw headers = X2

"Connect your inverters BMS RJ45 to the CAN IN RJ45 of the PCB."

CAN IN RJ45 = RJ1
On my Deye inverter this is the port labeled "BMS", right?

"If you want to connect your inverter and BMS via CAN. connect your BMS to the BMS IN RJ45 of the PCB."

BMS IN RJ45 = RJ2

Checking the PCB layout, this will just connect my BMS with the inverter and the PCB is just the "bridge" regarding cabling, right?

"I used RJ45 cables without shield connected to the RJ45 connector"

So no ethernet cable but some very simple RJ45 "phone" cable?

"Connect your Seplos BMS to RJ3"

So the R485 Port of my Battery is now connected to the PCB. But I still need the battery to be connected to my Deye inverter.
I got two options:

A: R485 is used via RJ1 (this option is true anyway...)
B: This is done via that "CAN bridge" (RJ1 -> RJ2) 

So B could be useful if, eg. the PCB board fails?

Pls correct my if I am wrong.

engel75, 2023-08-04:
Any idea where to get the U2 (HB PH8 20024PB2GOP)?
It is just for serial debugging, right? So I might leave it blank?

bagges, 2023-08-04:
> Any idea where to get the U2 (HB PH8 20024PB2GOP)? It is just for serial debugging, right? So I might leave it blank?

This is just for further extensions. you can leave it blank.

bagges, 2023-08-04:
> Thank you for adding this example!
>
> I am almost finished with the PCB board, but I would like to clarify some questions I have regarding your wiring description:
>
> "Connect any 7-24V DC source to the [...]

Yes

> "Connect your inverters BMS RJ45 to the CAN IN RJ45 of the PCB."
> 
> CAN IN RJ45 = RJ1 On my Deye inverter this is the port labeled "BMS", right? 

Yes

> "If you want to connect your inverter and BMS via CAN. connect your BMS to the BMS IN RJ45 of the PCB."
> 
> BMS IN RJ45 = RJ2

To BMS CAN, yes

> Checking the PCB layout, this will just connect my BMS with the inverter and the PCB is just the "bridge" regarding cabling, right?

Yes

> "I used RJ45 cables without shield connected to the RJ45 connector"
> 
> So no ethernet cable but some very simple RJ45 "phone" cable?

Also not all ethernet cables to have a shield. In the latest layout the shield of the rj45 jack is not connected, so shielded cable should also make no problems.

> "Connect your Seplos BMS to RJ3"
> 
> So the R485 Port of my Battery is now connected to the PCB. But I still need the battery to be connected to my Deye inverter. I got two options:
> 
> A: R485 is used via RJ1 (this option is true anyway...) B: This is done via that "CAN bridge" (RJ1 -> RJ2) 

You cannot connect your Deye via RS485 and this PCB.

> So B could be useful if, eg. the PCB board fails?

This is the only way to connect your BMS with Deye. Otherwise you have to use BatV/Bat% settings in Deye.

> Pls correct my if I am wrong.

engel75, 2023-08-09:
I finished the board and flashed the esp32. Looks promising right now. I am still waiting for my 12V power supply and the lifepo4 battery. I ordered your very nice looking housing as well, but the screws are driving me crazy :)
I was not able to find any screw that would fit. Any hints regarding the screws to close the housing?

bagges, 2023-08-09:
It is designed to fit those: https://www.amazon.de/gp/product/B0B51MFXZQ/

Then you can just use normal m3 screws.

engel75, 2023-08-09:
Ok I removed all the Pace BMS stuff to only log the Deye but I get:

`INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from esphome-web-4da3ec.local using esphome API
INFO Successfully connected to esphome-web-4da3ec.local
[23:22:38][I][app:102]: ESPHome version 2023.7.1 compiled on Aug  9 2023, 23:22:11
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=130 count=1
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=145 count=2
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=148 count=12
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=166 count=12
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=500 count=1
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=502 count=1
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=524 count=1
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=527 count=3
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=534 count=2
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=541 count=1
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=548 count=1
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=551 count=8
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=586 count=3
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=590 count=3
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=598 count=3
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=604 count=4
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=616 count=4
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=625 count=1
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=630 count=3
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=638 count=1
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=653 count=1
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=655 count=1
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=672 count=2
[23:22:42][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=676 count=4
[23:22:42][W][component:204]: Component modbus_controller took a long time for an operation (0.21 s).
[23:22:42][W][component:205]: Components should block for at most 20-30ms.
[23:22:50][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=65 count=1
[23:22:50][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=99 count=3
[23:22:50][W][modbus_controller:113]: Duplicate modbus command found: type=0x3 address=108 count=2`

Any idea?

bagges, 2023-08-09:
Bad wiring, no termination resistor, bad power supply.

engel75, 2023-08-10:
> Bad wiring, no termination resistor, bad power supply.

looks like the deye did not "save" the modbus SN01. It was still set to 00. Got all values now! Awesome!

bagges, 2023-08-10:
Does PaceBMS now also work?

Gisbert1, 2023-08-30:
Dear Markus,

I finalized the board yesterday evening, I measured 118 Ohm on RS485 boards, and started to connect the board to my Delongtop HS51200 battery and DEYE 12kW inverter. PACE_MODBUS was selectet with the BMS tool and the RS485-Port of the battery is connected to RJ3.In Advanced Settings in the DEYE, Modbus 01 Slave is selectet.

The communication of the board to the battery is working and the data are published via MQTT. So far so good.

When I plugged the BMS port (see photo with yellow cable) on the inverter either to RJ1 or RJ2, I got a communication error "W31 Battery_comm_warn". The battery is still loaded by the DEYE but there is no communication.

According to the DEYE approved battery list, the battery should be connected with the CAN port (at the battery) with inverter setup 00 - this is working fine.

What's about the other port in the DEYE named ModBUS? Shall I use this port for the connection to RJ1 or RJ2 and leave the cable Battery/CAN - DEYE/BMS where it is?

Regards
Gisbert

[image]

[image]

Gisbert1, 2023-08-30:
One more question, do I need to close the jumpers 1 and 4. Currently they are open.

Gisbert1, 2023-08-30:
Dear Markus,

thanks for your explanation, now it works.

For those who might have the same experience as me, the "RJ1 is to connect your Inverter to the board (reads data from the inverter over RS485)" which is in my case the BMS port of the inverter (see the picture above).

The second thing was that I gave up to early when the wiring was alright but the DEYE has still the communation warning and the warning signal. After a minute the warning stopped and everthing works normal.

Many Thanks
and Cheers
Gisbert