bagges/deye-esp32-bridge #4: MQTT

Gisbert1, 2023-08-09:
As far as I understand, the information is provided in a Webbrowser when typing the IP address of the ESP32.
I want the data in my house automatisation FHEM. Is it possible to publish the information via MQTT? Or is there any other way to provide the information to my house automatisation?

bagges, 2023-08-09:
For mqtt you have to provide mqtt config in the yaml. That should then work with fhem, iobroker,...

Gisbert1, 2023-08-09:
Hi bagges,
for me it's more than difficult, at my level even impossible, to add MQTT to your code (I'm using credentials with my MQTT broker).
Could you help me with the implementation and an example? If it's then necessary to put the code multiple times at different lines in your code, I could do that.
Regards and Greetings from NRW to Saarland
Gisbert

bagges, 2023-08-09:
You just have to add this one time to your yaml:

```
mqtt:
  broker: IP
  port: PORT
  topic_prefix: PREFIX (opt.)
  username: USER (opt.)
  password: PASSW (opt.)
```

Gisbert1, 2023-08-10:
Hello bagges,
MQTT works now with your project. The next steps for me are to solder everthing together and to add the board to the DEYE and the battery.
Many Thanks and Greetings
Gisbert

closed 2023-08-10 by bagges