bagges/deye-esp32-bridge #5: [feature] external fan control

engel75, 2023-08-14:
Hi,
what about adding support to control "external" 12V PWM fans to cool the Deye Inverter based on a temperature sensor? My Deye 12k is quite noisy on sunny days. To me, it sounds like those fans are not very well controlled. What about adding additional 12V PWM Fans (without opening the inverter!) to the bottom or top of the inverter, controlled by the ESP32 based on the temperature of the Deye case or cooling fins?

bagges, 2023-08-28:
Yeah good idea. Will think about adding such a feature.

bagges, 2023-09-08:
I checked this and it should already be possible with external 12V power supply. I ordered some fans, will try in the next days.