bagges/deye-esp32-bridge #7: RJ45 Jack

pollierer, 2023-09-13:
Awesome project, thank you for sharing!
I'm about to do my own version of a PCB and I found you use exactly the RJ45 jack I have at home in spare. I could not find any footprint for this parts, so I wanted to ask if you could provide your footprints. That would be great!
I use KiCad but this should be no issue.
When I finish the project I can share my experience with connecting my Deye Sun-8k..lp03..eu
Thank you!

bagges, 2023-09-13:
How does your own version differ? I used the easyeda model from this site: https://www.lcsc.com/product-detail/Ethernet-Connectors-Modular-Connectors-RJ45-RJ11_Shenzhen-Kinghelm-Elec-KH-RJ45-59-8P8C_C2683358.html

pollierer, 2023-09-14:
Thanks for the link @bagges ! :)
My Version will be more simple, I only need one rs485 and I want to add a downstepper module and my own power jack, also I want to add a ESP32 in D1 mini format, that´s the main reason I do my own one. Btw I like that you´re using jumpers, very versatile - will propably steal this idea for future projects ;)

closed 2023-09-14 by bagges