bagges/deye-esp32-bridge #8: no bug - but explanation for pcb

AndyMuc72, 2023-09-18:
Hello,

I love your project. I ordered the pcb and started soldering. I also checked for the 120 Ohm on the 485_2_TTLs.
But I don't understand what are the connectors U4 (Sensors written on the PCB) and what is U2? Is it a connector?
Is it possible to show also a picture of the finished soldered pcb?

thank you
Andy

bagges, 2023-09-18:
U4 is to connect for example a ds18b20 temperature sensor. U2 is just a connector for some I/O and Power/GND pins if you want to do some extensions (LCD for example).

Finished PCB I can send a picture afterwards.

AndyMuc72, 2023-09-18:
that makes sense. Where can I get this connector U2 in Germany and what is its name?
I have a Pylontech Battery, so I don't know if this project is useful for me. Did you think about implementing Pylontech?

Did you print a case?

bagges, 2023-09-18:
U2 is just for normal jumper wires, I do not know if there is a special connector for this. Pylontech has a different pinout on the rj45, this does not work.
Yes I printed the case from this repo.

AndyMuc72, 2023-09-28:
Ich habe gesehen, dass die Pylontech Akkus die RS485 Pins auf der anderen Seite haben. Wenn man ein spezielles Kabel macht, dann sollte man die Werte auch auslesen können, oder (und natürlich die Modbus Adressen anpassen).
Ich benutze den Adapter erstmal für den Deye.
Ich habe aber festgestellt, dass ich sehr viele CRC Fehler habe und auch ab und zu falsche Werte bekomme (z.B. "${device_type}-PV2 Power" liefert bei aktueller Leistung 2000W plötzlich einen Wert von 65000). Hier musste ich jetzt mit LAMBDA Filtern arbeiten, um die Werte zu ignorieren.
Hast du da irgendeine Idee, an was das liegen kann?

bagges, 2023-09-28:
Are you using an ESP32 or ESP8266?
CRC errors often occur because of bad power supply or bad bus termination.

AndyMuc72, 2023-09-28:
It is a ESP32. That is a good hint. I use an USB power supply and there is a Raspi 4 on this power supply too. I am not physically there right now. When I am there I will change it and see what will happen. Thank you

bagges, 2023-10-03:
Checkout the README, JP2 must not be closed when USB power connected. USB power supply often make problems.

AndyMuc72, 2023-10-03:
ok, thanx, I just read it, too. I will try a separate power supply next. all jumpers are open.

closed 2023-10-03 by bagges