bagges/deye-esp32-bridge #9: [Possible issue] DC-DC Converter inner pins not matching and Documentation request

xF4m3, 2023-11-03:
Hey, not sure if that is an issue in general orwith the DC-DC Converter that i received.

For me only the outer 4 holes are close to identical(not perfect), but the 4 inner holes of the Converter do not fit, i can only get them to show a tiny bit if i shift the board a bit so that all 8 holes atleast are visible in the slightes. I do know that its not needed since the 4 inner and 4 outer holes are just a "mirror" of each other. But it would be nice if that could be double checked. 

Of course it could also be diffrences in the DC-DC converters, but i used the ones that were linked in the BOM.

Additionally, i don't want to open a second issue just for that, but it would be nice if you could make a short description what each of the Jumpers do. Only JP2 is mentioned so far. maybe a small Table with all jumpers/connectors/headers and what they are for would be a nice addition.

bagges, 2023-11-03:
The DC-DC Converter I received, have a perfect fit with the PCB. Could you post a picture?

I updated the readme to contain a table with component description: https://github.com/bagges/deye-esp32-bridge/commit/7357dff866de68883b9de0463fbbb4cc4467fbf4

xF4m3, 2023-11-04:
I have already soldered everything and tried to align everything even if that means a hole is only visible 1/3, just tried to flush it with solder.
But i just made some photos. You can see that the distance from inner to outer is about 2.5mm while on the dc-dc converter its about 3mm(only a rough measurement, as seen in the photos).
 
[image]
[image]

bagges, 2023-11-04:
Really strange, mine do perfect fit RM2.54...