bagges/deye-esp32-bridge #14: Darstellung der Werte in HA

ThiloGa, 2023-12-12:
Moin,

dank der hier geleisteten Arbeit bekomme ich Werte in HA.

Allerdings bekomme ich bei PV Total Power keine Verlaufskurve, sondern
einen sich farblich verändernden Balken. 

[image]

An welche Schraube wäre hier zu drehen?
Folgefehler ist sicherlich, dass sich dieser Wert und der Outputpowerwert von OpenDTU per Helper nicht addieren lassen.

Danke

Grüsse

ThiloGa, 2024-02-12:
> Should work now

Better, but not... :-)

Now i can sum up the power values of deye und opendtu, but here...

[image]

... is nothing :-O

after click on "Verlauf" there is graph

[image]

ThiloGa, 2024-02-12:
> > Should work now
> 
> Better, but not... :-)

Further investigation shows, that PV[1|2] Power is working fine, so the problem must be buildiung the sum of PV[1|2] Power...

bagges, 2024-02-12:
Hm, I need to investigate to this. Maybe setting force_update to true could fix this.

commit 20817185 (2023-12-12): fixed #14