bagges/deye-esp32-bridge #16: Two PACE-BMS + Deye

m1k3f15h, 2024-03-25:
Hey Guys,

Did somebody tested this with 2 PACE-BMS (on one Modbus) + 1 Deye?
Is the ESP32 "Strong and Fast" enouch?

greetings from Austria
Mike

bagges, 2024-05-22:
Did you have success with this? I do not have PACE BMS, cannot test...

S-t-e-p-s-i, 2024-05-26:
Hello Mike I am also interested in, if you were successful.

@bagges really great board and project for the DEYE setup!! 
I started from scratch three weeks ago and knew I could work with many data points in HA. Really impressive. 

My setup: Deye Sun10k/FM-Solar stackable 2 × 10 kWh battery storage.

S-t-e-p-s-i, 2024-06-03:
Maybe I can place a question about the possible setup here...

I get all the relevant data from the Deye and Pace BMS via the bridge, works without any problems so far.

In order to read the values from the second battery pack, I have looped the RS485 through to the bridge. I can access the data of the respective battery banks individually via the address 0x01 / 0x02, but I can't manage to process the data for both banks in the Yaml code via a reciver/converter.

With regard to master and slave processing of the converters, it seems to me that these are limited in this technical function?
Or may i understood the coding not fully ... 

https://esphome.io/components/modbus_controller.html

would then mean that I need a separate converter for the second battery bank.
If someone can give a helpful tip whether it can be processed with a corresponding adjustment in the Yaml would be really great.

Greetings Stefan 

esphome:
  name: deye-bridge
   
esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

# Enable logging
logger:
  level: INFO

substitutions:
  bms_name_1: Batterypack1
  bms_name_2: Batterypack2
  modbus_update_interval: "8s"
  settings_skipped_updates: "10"
  query_throttle: "50ms"
  device_name: deye_bridge
  device_type: sun10k_bridge
  device_description: "Esphome component for Deye sun-10k-sg04lp3"
  modbus_controller_id: sg04lp3

# Enable Home Assistant API
api:
  encryption:
    key: xxxx

ota:
  password: xxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Deye-Bridge Fallback Hotspot"
    password: xxx

captive_portal:

uart:
  - id: uartdeye
    tx_pin: 27
    rx_pin: 26
    baud_rate: 9600
  - id: uartpace
    tx_pin: 17
    rx_pin: 16
    baud_rate: 9600
    debug:
      direction: BOTH
      dummy_receiver: false

modbus_controller:
  - id: ${modbus_controller_id}
    address: 0x01
    modbus_id: modbus1
    setup_priority: -10
    update_interval: ${modbus_update_interval}
    command_throttle: ${query_throttle}
  - id: bms0
    # Slave address 0x01
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 6s
# - id: bms1
#    # Slave address 0x02
    #address: 0x02
    #modbus_id: modbus2
    #command_throttle: 300ms
    #update_interval: 8s

modbus:
  - id: modbus1
    uart_id: uartdeye
  - id: modbus0
    uart_id: uartpace
    send_wait_time: 200ms
  #- id: modbus2
  #  uart_id: uartpace
  #  send_wait_time: 300ms
time:
  - platform: sntp
    id: sntp_time
    timezone: Europe/Berlin
    servers:
    - pool.ntp.org
    
Sensor:      #(eg. how i tried it.....)

  #  15  Cell voltage 1                        2 byte   R  uint16  mV
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${bms_name_1} cell voltage 1"
    address: 15
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    state_class: measurement
    accuracy_decimals: 3
    filters:
      - multiply: 0.001
  - platform: modbus_controller
    modbus_controller_id: bms1
    name: "${bms_name_2} cell voltage 1"
    address: 15
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    state_class: measurement
    accuracy_decimals: 3
    filters:
      - multiply: 0.001

m1k3f15h, 2024-06-06:
Hy Stefan

This is my YML for 1 Deye and 2 Batterys.. maybe this helps you.
[ess.txt](https://github.com/user-attachments/files/15605577/ess.txt)

Mike

bagges, 2025-03-11:
closing because no more information was provided.

closed 2025-03-11 by bagges