bagges/deye-esp32-bridge #19: ESP32 hängt sich nach ein paar Tagen auf

Lembi203, 2024-05-23:
Nach ein paar Tagen ( 1-7 ) werden die Entitäten unavailable und der ESP geht offline, ich habe genau die verlinkten Komponenten bestellt, aber es läuft nicht zuverlässig.

Logs hab ich keine weil es so unregelmäßig passiert und der ESP offline geht.

Ich habe schon mehrere ESP vom gleichen Typ an Modbus Zählern usw, das funktioniert zuverlässig.

bagges, 2024-05-23:
Welche ESPHome Version?
Auch das gleiche Netzteil verwendet?

bagges, 2024-05-28:
Ohne Logs wird das echt schwierig. Kannst du die Netzteile mal verlinken? Über USB geht nicht, der Spannungswandler bringt nicht genug für alle Komponenten.

m1k3f15h, 2024-06-07:
Was für einen ESP / Board hast du?
mike

Lembi203, 2024-06-07:
Das verlinkte von AZDelivery

m1k3f15h, 2024-06-07:
wo verlinkt?

Lembi203, 2024-06-07:
Den der in der Readme verlinkt ist. 
[AZDelivery ESP32 Dev v4](https://amzn.to/41Sd0v4)

m1k3f15h, 2024-06-07:
okay.. da läuft es bei mir jetzt zum testen...

bei dem BOARD: [https://www.amazon.de/AZDelivery-Bluetooth-Internet-Entwicklungsboard-kompatibel/dp/B08BTLYSTM/ref=sr_1_1_sspa?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=12JWCDISUGCD4&dib=eyJ2IjoiMSJ9.Hvygg_Q0VE0DxzuMVicAQ9J_lyGStIBIF-GbBAh0B8aYsk3wDPe98iRuTP_AnjxHbUmbTmPRWqSEsfAnDCFGrGx3qqZpl1ldwNtv66H9BzmA9F62fO1TAc9nZe9M1IAnMoik1vMy0daiT4c1eTG2eD3efaYXYwIub3jQQBPCjVbj-ZC6CRTc0N3ssHvYKnJlVmMqVnjU2aTBpH3Oa2ZWPCorEg7pmlXSBKkdk9JD2D0.iJr6R8HVnquzjmW7b2u4ug0FYlLgDb_vlQbjubSf_Xw&dib_tag=se&keywords=azdelivery+esp32+wemos&qid=1717771991&sprefix=az+delivery+esp32+wemos%2Caps%2C91&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1](url) hab ich das problem, dass der alle paar tage aufgibt...

sven-s, 2024-09-07:
Ich habe auch das Problem, dass sich das Board immer wieder offline meldet. Hast du den Kondensator an die Versorgungsleitung des ESP32 gehängt? Also an Plus und Masse?

bagges, 2024-09-08:
> Ich habe auch das Problem, dass sich das Board immer wieder offline meldet. Hast du den Kondensator an die Versorgungsleitung des ESP32 gehängt? Also an Plus und Masse?

Auf dem Step Down sind Kondensatoren drauf. Ich habe sonst nichts zusätzlich. Wie versorgst du denn mit Strom? Ich hatte das Problem, wenn ich entweder direkt über den ESP gehe (der Spannungsregler liefert nicht genug Saft), oder ein Netzteil mit zu wenig Saft an dem DCDC hatte.

sven-s, 2024-09-08:
Hatte zuerst eine 15 Watt USB Netzteil, nun aber das Poppstar Netzteil für Hutschiene 12V DC 1,25A 15 W (Input 100-240 V AC 50/60 Hz). Habe noch einen Elko direkt vor den Step Down gehängt, aber leider kein Erfolg. 

Bei mir zeigt sich das folgendermassen: 

- Ping erfolgt nicht durchgehend, und auch die UI ist zeitweise nicht erreichbar.
- Das MQTT Topic geht auf Status Offline

Das Log in der UI sieht so aus:

[image]

Kann ich noch weitere Logs auslesen? Kann der ESP evtl. zu heiss werden und daher offline gehen bzw. neustarten?

bagges, 2024-09-08:
Ping erfolgt nicht durchgehen: hast du einfach schlechtes WLAN? Es scheint auch was mit deiner Modbus Verbindung nicht zu stimmen. Die crc Fehler habe ich nicht.

sven-s, 2024-09-08:
Habe in der Garage einen eigenen Unifi AP. Der ist nur 3 m Luftlinie entfernt, von daher würde das erstmal ausschließen.

bagges, 2024-09-08:
Check trotzdem mal ob der ESP mit diesem verbunden ist. Hatte schon ähnliche Konstellation und musste den ESP auf einen AP locken...

sven-s, 2024-09-08:
Ja, der ist mit dem AP verbunden mit -46dBm. Habe den jetzt auch gelockt. Ping kommt aber nicht durchgehend. Werde mal mit der aktuellsten esphome Version flashen.

sven-s, 2024-09-08:
Habe auch noch einen anderen ESP an dem Unifi AP per Wifi. Der lässt sich durchgehend anpingen... das muss an etwas anderem liegen.

m1k3f15h, 2024-09-09:
> Ich habe auch das Problem, dass sich das Board immer wieder offline meldet. Hast du den Kondensator an die Versorgungsleitung des ESP32 gehängt? Also an Plus und Masse?

Servus,

Ja ich hab den Kondensator so nahe wie möglich am DevBoard an die Versorgungsspannung gelötet.. seit dem keine Probleme mehr :)

sven-s, 2024-09-10:
So, habe nochmals alles gecheckt, Kabel getauscht, Widerstand vom RS485 gemessen und so weiter. Hat alles nicht geholfen. 

Über das Log habe ich immer wieder die Meldung gehabt:

```
[20:44:52][I][app:100]: ESPHome version 2024.8.3 compiled on Sep  8 2024, 23:20:30
[20:44:57]E (28403) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[20:44:57]E (28403) task_wdt:  - loopTask (CPU 1)
[20:44:57]E (28403) task_wdt: Tasks currently running:
[20:44:57]E (28403) task_wdt: CPU 0: IDLE
[20:44:57]E (28403) task_wdt: CPU 1: IDLE
[20:44:57]E (28403) task_wdt: Aborting.
[20:44:57]
[20:44:57]abort() was called at PC 0x4010bbbc on core 0
[20:44:57]
[20:44:57]
[20:44:57]Backtrace:0x400838f9:0x3ffbe9cc |<-CORRUPTED
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400838f9: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
[20:44:57]
[20:44:57]
[20:44:57]
[20:44:57]
[20:44:57]ELF file SHA256: 0000000000000000
[20:44:57]
[20:44:57]Rebooting...
[20:44:57]ets Jul 29 2019 12:21:46
[20:44:57]
[20:44:57]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[20:44:57]configsip: 0, SPIWP:0xee
[20:44:57]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[20:44:57]mode:DIO, clock div:2
[20:44:57]load:0x3fff0030,len:1184
[20:44:57]load:0x40078000,len:13132
[20:44:57]load:0x40080400,len:3036
[20:44:57]entry 0x400805e4
```

Ich nutze die PACE yaml, da ich einen Deye mit Gobel Akku habe. 

Dann habe ich mal einige Configs (ca. 20) aus dem Pace Bereich auskommentiert, wie:

```
  # # 100  Environment OT protection             2 byte  RW   int16  0.1 ℃
  # - platform: modbus_controller
  #   modbus_controller_id: bms0
  #   name: "${bms_name} environment overtemperature protection"
  #   use_write_multiple: true
  #   address: 100
  #   register_type: holding
  #   value_type: S_WORD
  #   lambda: "return  x * 0.1f; "
  #   write_lambda: "return x * 10.0f;"
  #   min_value: -200.0
  #   max_value: 200.0
  #   step: 0.1
  #   mode: box
  #   unit_of_measurement: "°C"
```

Seitdem läuft das deutlich besser. Die Werte kommen durchgehend und der Ping ist auch die ganze Zeit da. 

Von daher gehe ich mal davon aus, dass der ESP mit der Anzahl der Register nicht zurechtgekommen ist.

bagges, 2024-09-10:
Der PACE Hinweis ist nicht ganz unwichtig -.-. Ich hab leider kein PACE und kann das nicht verifizieren. Statt Configs rauszunehmen kannst du auch versuchen das ```modbus_update_interval``` höher zu setzen.

bagges, 2025-03-11:
Ich schließe das Issue hier mal. Falls es noch Probleme gibt bitte nochmal schreiben.

closed 2025-03-11 by bagges