bagges/deye-esp32-bridge #20: Gehäuse, welche Schrauben sind da vorgesehen?

ThiloGa, 2024-06-01:
Kein Code issue... :-)

welche Schrauben sollen eigentlich das Case zusammenhalten?

Hab in meiner grossen Schraubenkiste nix passendes gefunden... ;-)

Grüsse
Thilo

S-t-e-p-s-i, 2024-06-02:
Hallo Thilo,

nach meinem Verständnis kannst du da Gewindebuchsen einpressen schau mal bei Amazon und Co nach Ruthex Gewindebuchsen für 3D Druck. So mache ich das zumindest bei meinen Teilen die auch länger halten sollen.

[ruthex](https://www.ruthex.de/)

Gruß Stefan

bagges, 2024-06-02:
Wie Stefan schon sagt. Ich nutze diese: https://amzn.eu/d/4xWvvk3 und schmelze sie mit dem Lötkolben ein.

closed 2024-06-02 by bagges

ThiloGa, 2024-06-04:
Hi Stefan,

> nach meinem Verständnis kannst du da Gewindebuchsen einpressen schau mal bei Amazon und Co nach Ruthex Gewindebuchsen für 3D Druck. So mache ich das zumindest bei meinen Teilen die auch länger halten sollen.

Danke für die Infos. Was zum 3D-Druck gelernt.

> 
> [ruthex](https://www.ruthex.de/)

Jetzt gibts da nur so grosse PE, brauche nur die 4 Stück.

Gibts ne Möglichkeit, dass du mir ein paar gegen Unkostenerstattung in einen Umschlag steckst?

Grüsse
Thilo

bagges, 2024-06-04:
@ThiloGa Schick mir n Euro über den Donate Button und ich schick dir n paar zu. Schreib deine Adresse ins Kommentarfeld.