bagges/deye-esp32-bridge #23: Summer winter time changeover

PVAnfaenger, 2024-08-02:
Hello everyone,
since the last time conversion still have an open point!
And that is that the Deye did not do this automatically. Now the question, can this be controlled via the Modbus? From address 62-64 the time is in the communication protocol. However, I don't know how to do this, how to combine 6 bytes in esphome into one variable and then describe it correctly.

Can someone help me here?

bagges, 2024-09-09:
As a starting point, here is how you can read the sensor:

`text_sensor:
  - platform: modbus_controller
    modbus_controller_id: ${modbus_controller_id}
    name: "${device_type}-system time"
    id: ${device_type}_system_time
    register_type: holding
    address: 62
    register_count: 3
    response_size: 6
    lambda: |-
      int jahr = (int)data[item->offset]+2000;
      int monat = (int)data[item->offset + 1];
      int tag = (int)data[item->offset + 2];
      int stunde = (int)data[item->offset + 3];
      int minute = (int)data[item->offset + 4];
      int sekunde = (int)data[item->offset + 5];
      char buffer[20];
      sprintf(buffer, "%d-%d-%d %d:%d:%d", jahr, monat, tag, stunde, minute, sekunde);
      return {buffer};`

Gives you:

[image]

Gisbert1, 2024-09-11:
Hi bagges,

your code works well and gives the current time of the DEYE.
I tried to search for a code to write a new date/time to the DEYE but my knowledge is by far too low.
To get always 4 or 2 digest for the date/time, the code could be:
...

PVAnfaenger, 2024-10-24:
I would be very interested in that too.

geronet1, 2025-02-13:
Solution, you can press the time sync button in HA or use the on_time logic:

```
substitutions:
  esphome_name: deye                                                 #name in ESPhome
	modbus_controller_id: sg04lp3

time:
  - platform: homeassistant
    id: esptime
  - platform: sntp
    on_time:
      - seconds: 0
        minutes: 0
        hours: 6
        days_of_week: MON
        then:
          - button.press: ${esphome_name}_time_sync

modbus:
  id: modbus1
# flow_control_pin: 4 #for use when you use a rs485 board without auto flow control. Like the rs485 max board.

modbus_controller:
  - id: ${modbus_controller_id}
    address: 0x1
    modbus_id: modbus1
    setup_priority: -10
    update_interval: ${intervall}
    #command_throttle: "50ms"

button:
  - platform: template
    name: ${name} time sync
    id: ${esphome_name}_time_sync
    icon: "mdi:clock-outline"
    on_press:
      then:
        - lambda: |-
            esphome::modbus_controller::ModbusController *controller = id(sg04lp3);

            time_t now = ::time(nullptr);
            struct tm *time_info = ::localtime(&now);
            int seconds = time_info->tm_sec;
            int minutes = time_info->tm_min;
            int hour = time_info->tm_hour;
            int day = time_info->tm_mday;
            int month = time_info->tm_mon + 1;
            int year = time_info->tm_year + 1900; /* The number of years since 1900   */

            // if there is no internet connection localtime returns year 70
            if (year != 1970)
            {
              std::vector<uint16_t> rtc_data = {uint16_t(((year-2000) << 8) | month), uint16_t((day << 8) | hour), uint16_t((minutes << 8) | seconds)};
              // Create a Modbus command item with the time information as the payload
              esphome::modbus_controller::ModbusCommandItem set_rtc_command =
                esphome::modbus_controller::ModbusCommandItem::create_write_multiple_command(controller, 62, rtc_data.size(), rtc_data);
              // Submit the command to the send queue
              controller->queue_command(set_rtc_command);
              ESP_LOGI("ModbusLambda", "Deye RTC set to %02d:%02d:%02d %02d.%02d.%04d", hour, minutes, seconds, day, month, year);
            }
            else
              ESP_LOGW("ModbusLambda", "RTC sync failed");

```