bagges/deye-esp32-bridge #25: [BatteryMode] maximum battery discharge current not always reset

bagges, 2024-11-14:
From time to time the battery discharge current is not reset to the previous value

tingeltanglbob, 2024-12-26:
I had the problem that although the Fix25 switched to Hold mode, it continued drawing 100A from the battery. I changed the code as shown in the file

[deye_esp32_bridge fix25 changed.txt](https://github.com/user-attachments/files/18254127/deye_esp32_bridge.fix25.changed.txt)

bagges, 2024-12-26:
@tingeltanglbob you just swaped lines [171](https://github.com/bagges/deye-esp32-bridge/blob/70bf9b45dd3bea3daef79f55c63ee801dec437eb/deye-esp32-bridge.yaml#L171) and 172 right? which makes totally sense, but I cannot belive that this fixes the issue...

closed 2026-03-01 by bagges